unosquare/raspberryio #217: sudo nuget install Unosquare.Raspberry.IO dont work anymore?

JanZumdick, 2020-10-13:
sudo nuget install Unosquare.Raspberry.IO dont work anymore? 
"An error occurred while retrieving package metadata for 'Unosquare.Raspberry.IO' from source 'nuget.org'.

 Index was outside the bounds of the array.

JanZumdick, 2020-10-13:
[image]

geoperez, 2020-10-13:
It looks like an issue with the nuget client. Can you try on another computer?

JanZumdick, 2020-10-13:
i did. 

On Raspberry it dont work (Testet on 2 Raspberry Pi 4 [kali and raspbian])
On my Windows-Computer, it work fine (Visual Studio 2019 Ent).

JanZumdick, 2020-10-13:
[image]
Still the same

closed 2020-12-20 by stale[bot]