unosquare/raspberryio #220: Error tryng to comunicate with I2C device

DanieleNeo, 2020-10-18:
**Describe the bug**
I don't know if it's a bug, surely I need some kind of support. I'm writing C# code for using INA229 device (voltage and current sensor). The sensor is correctly initialized and seen by my single board computer

```
root@orangepipc:~# i2cdetect -y 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: 40 -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

The code I wrote (it's a draft for now) is this:
```
using System;
using System.Collections.Generic;
using System.Text;
using Unosquare.RaspberryIO;
using Unosquare.RaspberryIO.Abstractions;

namespace NS.HA.Hardware.VoltageAndCurrent
{
    public class INA226 : IDisposable 
    {

        private enum REG
        {
            INA226_REG_CONFIGURATION = 0x00,
            INA226_REG_SHUNT_VOLTAGE = 0x01,
            INA226_REG_BUS_VOLTAGE = 0x02,
            INA226_REG_POWER = 0x03,
            INA226_REG_CURRENT = 0x04,
            INA226_REG_CALIBRATION = 0x05,
            INA226_REG_MASK_ENABLE = 0x06,
            INA226_REG_ALERT_LIMIT = 0x07,
            INA226_REG_MANUFACTURER = 0xFE,
            INA226_REG_DIE_ID = 0xFF
        }

        private enum BITS
        {
            INA226_BIT_SHUNT = 0,
            INA226_BIT_BUS = 1,
            INA226_BIT_MODE = 2
        }

        private enum MODE
        {
            INA226_MODE_OFF = 0,
            INA226_MODE_SHUNT_TRIGGERED = 1,
            INA226_MODE_BUS_TRIGGERED = 2,
            INA226_MODE_SHUNT_BUS_TRIGGERED = 3,
            INA226_MODE_OFF2 = 4,
            INA226_MODE_SHUNT_CONTINUOUS = 5,
            INA226_MODE_BUS_CONTINUOUS = 6,
            INA226_MODE_SHUNT_BUS_CONTINUOUS = 7
        }

        private enum TIME
        {
            INA226_TIME_01MS = 0, // 140us
            INA226_TIME_02MS = 1, // 204us
            INA226_TIME_03MS = 2, // 332us
            INA226_TIME_05MS = 3, // 588us
            INA226_TIME_1MS = 4, // 1.1ms
            INA226_TIME_2MS = 5, // 2.115ms
            INA226_TIME_4MS = 6, // 4.156ms
            INA226_TIME_8MS = 7 // 8.244ms
        }

        private enum AVERAGES
        {
            INA226_AVERAGES_1 = 0,
            INA226_AVERAGES_4 = 1,
            INA226_AVERAGES_16 = 2,
            INA226_AVERAGES_64 = 3,
            INA226_AVERAGES_128 = 4,
            INA226_AVERAGES_256 = 5,
            INA226_AVERAGES_512 = 6,
            INA226_AVERAGES_1024 = 7
        }

        const int INA226_RESET = 0x8000;
        const int INA226_MASK_ENABLE_CVRF = 0x0008;
        const int INA226_MODE_SHUNT = 1;
        const int INA226_MODE_BUS = 2;
        const int INA226_MODE_TRIGGERED = 0;
        const int INA226_MODE_CONTINUOUS = 4;
        const int INA226_ADDRESS = 0x40;

        int fd;
        uint config;
        double current_lsb;

        double voltage, current, power, shunt, energy, price;

        II2CDevice _Device = null;
        II2CBus _Bus = null;

        public INA226()
        {
            //_Bus = new Unosquare.WiringPi.I2CBus();
            _Device = Pi.I2C.AddDevice(INA226_ADDRESS); //_Bus.GetDeviceById(INA226_ADDRESS);

#if DEBUG
            NS.HA.Logger.Logger.Log(Logger.Logger.eLog.DEBUG, "Manifacturer {0} chip {1} ",Read((int)REG.INA226_REG_MANUFACTURER),Read((int)REG.INA226_REG_DIE_ID));
#endif

            // Calibrazione
            Calibrate(0.1, 10);

            //Configurazione
            Configure(TIME.INA226_TIME_8MS, TIME.INA226_TIME_8MS, AVERAGES.INA226_AVERAGES_16, MODE.INA226_MODE_SHUNT_BUS_CONTINUOUS);

        }

        private void Calibrate(double Shunt, double MaxCurrent)
        {
            current_lsb = MaxCurrent / (1 << 15);
            var calib = 0.00512 / (current_lsb * Shunt);
            var calib_reg = Math.Floor(calib);
            current_lsb = 0.00512 / (Shunt * calib_reg);

            Write((int)REG.INA226_REG_CALIBRATION, (ushort)calib_reg);
        }

        private void Configure(TIME BusTime, TIME ShuntTime, AVERAGES Averages, MODE Mode)
        {
            config = ((uint)Averages << 9) | ((uint)BusTime << 6) | ((uint)ShuntTime << 3) | (uint)Mode;
            Write((int)REG.INA226_REG_CONFIGURATION, (ushort)config);
        }

        public void ReadINA()
        {

            // Legge il voltaggio
            var voltage_reg = Read((int)REG.INA226_REG_BUS_VOLTAGE);
            voltage = voltage_reg * 1.25e-3;

            // Legge la corrente
            var current_reg = Read((int)REG.INA226_REG_CURRENT);
            current = current_reg * 1000.0 * current_lsb;

            // Legge la potenza
            var power_reg = Read((int)REG.INA226_REG_POWER);
            power = power_reg * 25000.0 * current_lsb;

            var shunt_voltage_reg = Read((int)REG.INA226_REG_SHUNT_VOLTAGE);
            shunt = shunt_voltage_reg * 2.5e-3;

        }

        public double Voltage
        {
            get
            {
                return voltage;
            }
        }

        public double Current
        {
            get
            {
                return current;
            }
        }

        private double Read(int addr)
        {
            var result=_Device.ReadAddressWord(addr);
            return (result << 8) | (result >> 8);
        }

        private void Write(int addr, ushort value)
        {
            _Device.WriteAddressWord(addr, (ushort)((value << 8) | (value >> 8)));
        }

        private void Disable()
        {
            Write((int)REG.INA226_REG_CONFIGURATION, INA226_RESET);
        }

        private void Reset()
        {
            Write((int)REG.INA226_REG_CONFIGURATION, (int)MODE.INA226_MODE_OFF);
        }

        public void Dispose()
        {
            Disable();
        }
    }
}
```

I was inspired by this project
https://github.com/MarioAriasGa/raspberry-pi-ina226

The problem is that when I try to read or write on device an exception occurred. This is the error I got for example trying to read manufacturer and chips information as in ina226.c file, using I2CDevice.ReadAddressWord method (it's the first thing before calibration and configuration, and I can assure that built ina226.c works and read values as well).

```
Message: A hardware exception occurred. Error Code: 6
ExtendedMesage: No such device or address
```

I checked read/write register and values, and are the same of ina226.c. The problem seems to be related to device initialization, but this should be quite simple with Unosquare lib...

`Pi.I2C.AddDevice(INA226_ADDRESS)`

Really I don't know what I can do, help would be appreciated.

**To Reproduce**
Steps are simple, but you need to have INA229 device. Inizialize it and try to read 0xFE register...

**Expected behavior**
After calibration and configuration I should be able to read voltage and current values as ina226.c does

**Devices**
OrangePI PC, INA226 device.

DanieleNeo, 2020-10-18:
I'm starting to think that OrangePI is not supported by Unosquare libraries. I have tried to read values from a DHT22 sensor and I'have no data, so I tried to read a simple high value on a GPIO pin, with no luck. The GPIO I tried to read was BCM 20

```
root@orangepipc:~# gpio readall
 +-----+-----+----------+------+---+-Orange Pi+---+---+------+---------+-----+--+
 | BCM | wPi |   Name   | Mode | V | Physical | V | Mode | Name     | wPi | BCM |
 +-----+-----+----------+------+---+----++----+---+------+----------+-----+-----+
 |     |     |     3.3v |      |   |  1 || 2  |   |      | 5v       |     |     |
 |  12 |   8 |    SDA.0 |   IN | 1 |  3 || 4  |   |      | 5V       |     |     |
 |  11 |   9 |    SCL.0 |   IN | 1 |  5 || 6  |   |      | 0v       |     |     |
 |   6 |   7 |   GPIO.7 |   IN | 0 |  7 || 8  | 0 | IN   | TxD3     | 15  | 13  |
 |     |     |       0v |      |   |  9 || 10 | 0 | IN   | RxD3     | 16  | 14  |
 |   1 |   0 |     RxD2 |   IN | 0 | 11 || 12 | 0 | IN   | GPIO.1   | 1   | 110 |
 |   0 |   2 |     TxD2 |   IN | 1 | 13 || 14 |   |      | 0v       |     |     |
 |   3 |   3 |     CTS2 |   IN | 0 | 15 || 16 | 0 | IN   | GPIO.4   | 4   | 68  |
 |     |     |     3.3v |      |   | 17 || 18 | 0 | IN   | GPIO.5   | 5   | 71  |
 |  64 |  12 |     MOSI |   IN | 0 | 19 || 20 |   |      | 0v       |     |     |
 |  65 |  13 |     MISO |   IN | 0 | 21 || 22 | 0 | IN   | RTS2     | 6   | 2   |
 |  66 |  14 |     SCLK |   IN | 0 | 23 || 24 | 0 | IN   | CE0      | 10  | 67  |
 |     |     |       0v |      |   | 25 || 26 | 0 | IN   | GPIO.11  | 11  | 21  |
 |  19 |  30 |    SDA.1 |   IN | 1 | 27 || 28 | 1 | IN   | SCL.1    | 31  | 18  |
 |   7 |  21 |  GPIO.21 |   IN | 1 | 29 || 30 |   |      | 0v       |     |     |
 |   8 |  22 |  GPIO.22 |   IN | 1 | 31 || 32 | 0 | IN   | RTS1     | 26  | 200 |
 |   9 |  23 |  GPIO.23 |   IN | 0 | 33 || 34 |   |      | 0v       |     |     |
 |  10 |  24 |  GPIO.24 |   IN | 0 | 35 || 36 | 0 | IN   | CTS1     | 27  | 201 |
 |  20 |  25 |  GPIO.25 |   IN | 1 | 37 || 38 | 0 | IN   | TxD1     | 28  | 198 |
 |     |     |       0v |      |   | 39 || 40 | 0 | IN   | RxD1     | 29  | 199 |
 +-----+-----+----------+------+---+----++----+---+------+----------+-----+-----+
 | BCM | wPi |   Name   | Mode | V | Physical | V | Mode | Name     | wPi | BCM |
 +-----+-----+----------+------+---+-Orange Pi+---+------+----------+-----+-----+
```
Is it possible to extend compatibility also to Orange PI PC hardware? Using C and python scripts on this board I'm able to read DS18B20, DHT22, INA226 and other devices with no problem. It would be nice if it was possible also using your libraries because I wrote an home automation system and and I wish I could make it work also on Orange PI hardware.

Pi.Gpio return a list of gpio but probably there is something wrong addressing the correct ones...

DanieleNeo, 2020-10-19:
The Orange PI hardware is quite similar to the Raspberry, I use it often because it is cheaper and has some hardware features (such as integrated microphone) useful for particular usage scenarios (for example to build radio gateways with telemetry support), without the need to use third party usb sound cards, which often introduce an element of weakness on system stability.

Having the hardware I remain at full disposal in case I have to do some tests, thanks in advance

D.

DanieleNeo, 2020-10-19:
I tried with Raspberry, and it's working. So, as I supposed, the problem is because Unosquare is not compatible with Orange PI hardware. So now I renew my request: it could be possible to extend support also to Orange PI hardware? Probably only little customizations are required, after all ina226.c is working and it's using wiringPiI2C...

Thanks in advance,
D.

DanieleNeo, 2020-10-21:
Dear @geoperez and @k3z0, if I'm here obviously I'm an admirer of your library, that I used before in my personal projects. I'm sure now, after doing several tests, that the problem is unsupported hardware issue. The ambiguous thing, which made me waste some time, is that on OrangePI hardware I'm able to see Pi.Gpio, but when I try to read gpio values I got always False (no warning or exception), while the problem presents itself more explicitly trying, for example, to write on i2c device. 

I would like to know a few things: if you think it is difficult to extend the library support to the orangepi as well (since with C/Python I can do basically the same things I do with a raspberry) and in case what would need to be done, from a technical point of view. Also if I can help you in anything, although my low level skills are not many.

Thanks in advance,
D.

geoperez, 2020-10-21:
Hi DanieleNeo, I like your idea to provide support to OrangePI but we are busy with other projects and we have no spare time to deliver a solution in the near future.

@k3z0 do you think you can help?

DanieleNeo, 2020-10-23:
Thank you @geoperez for you reply, I understand your motives very well. I hope that @k3z0 can help us in this activity, unfortunately I don't have the necessary skills, otherwise I would try to do something myself. Is there a point in the code where i can look?

closed 2021-01-03 by stale[bot]