unosquare/raspberryio #222: Unable to load shared library 'libwiringPi.so.2.52

callmenoob, 2020-10-25:
Hi, im using Raspberry PI 4B Ubuntu 19.10 (arm64). When I'm trying to launch bootstrap procces I'm getting error 
` Unable to load shared library 'libwiringPi.so.2.52' or one of its dependencies.` 
How can i get this dll? I have tried this guide 
http://wiringpi.com/wiringpi-updated-to-2-52-for-the-raspberry-pi-4b/
but this package is not compateble with arm64, is there a workaround for this problem?

wolfgangmauer, 2020-10-25:
I just have a Pi3B+...

What "ls" shown? I should look something like this

pi@raspberrypi:/tmp $ **ls -l /usr/lib/libwiringPi***
lrwxrwxrwx 1 root root    19 Okt 25 15:00 /usr/lib/libwiringPi.so -> libwiringPi.so.2.52
-rwxr-xr-x 1 root root 69808 Jun 23  2019 /usr/lib/libwiringPi.so.2.52

mfertala, 2020-12-17:
Hi! I experience the same error on RasPi 4B Ubuntu 20.04.1 (arm64). When code comes to:
`Pi.Init<BootstrapWiringPi>();`
I get error:
`Unhandled exception. System.DllNotFoundException: Unable to load shared library 'libwiringPi.so.2.52' or one of its dependencies. In order to help diagnose loading problems, consider setting the LD_DEBUG environment variable: liblibwiringPi.so.2.52: cannot open shared object file: No such file or directory
   at Unosquare.WiringPi.Native.WiringPi.WiringPiSetupGpio()
   at Unosquare.WiringPi.GpioController.Initialize(ControllerMode mode)
   at Unosquare.WiringPi.GpioController..ctor()
   at Unosquare.WiringPi.BootstrapWiringPi.Bootstrap()
   at Unosquare.RaspberryIO.Pi.Init[T]()
   at RPI_GPIO_Tests.PiGpioHepler..ctor() in /home/mf/GIT/raspberryhome/RPI_GPIO_Tests/PiGpioHepler.cs:line 12
   at RPI_GPIO_Tests.Program.Main(String[] args) in /home/mf/GIT/raspberryhome/RPI_GPIO_Tests/Program.cs:line 26`

geoperez, 2021-02-15:
We are looking for a new maintainer, apply at https://adoptoposs.org/p/9f5b74b9-04f2-42b6-891f-c5294c9ef1c5

closed 2021-06-16 by stale[bot]