unosquare/raspberryio #226: wiringPiISR: Can't find gpio program

IronSean, 2020-11-10:
**Describe the bug**
Hi, I'm working on an implementation with raspberryio and I'm getting this error when running:
`wiringPiISR: Can't find gpio program`

I'm using `Pi.Init<BootstrapWiringPi>();` in my program.cs, and I've got the Unosquare.WiringPi package installed too.

I'm also building and running on a docker container, I'm not sure if that is related.

**To Reproduce**
Steps to reproduce the behavior:
1.Start app
2. Init Pi
3. Init and add interrupt to GPIO pin

**Expected behavior**
Interrupts fired by GPIO pin rising edge

**Devices**
Pi 4

IronSean, 2020-11-10:
I've ensured I'm on the latest Unosquare.WiringPi v0.51, which should support the Pi 4b. 

I also tested Unosquare.PiGpio because WiringPi is apparently deprecated by it's maintainer. Unfortunately, none of the GPIO methods are implemented yet on the PiGpio implementation.

I've not yet tried running the app natively on the Pi, as getting dotnet core running in docker was easier than installing it natively, and docker is how we intend to eventually deploy this applicaiton to other Pis. 

However, I might need to test this in order to determine if it's an issue with docker or my application code itself.

IronSean, 2020-11-10:
I'm not experiencing the error when run natively instead of within Docker, so I think the real question is "How do we forward the right ports/files/etc. to allow Unosquare.WiringPi to run in a docker container and register gpio interrupts?"

IronSean, 2020-11-17:
To follow up, I was also running with `--privileged` set, and I believe I tried mounting the `/sys` folder as a volume. 

But also to follow up, I've moved on to solving this using a native C application to handle my GPIO interactions, and am no longer actively working on testing or debugging this issue. As such I may not be able to test suggestions in the future. I'll leave it up to the maintainers whether to keep this open or close until someone else experiences the issue.

closed 2021-01-25 by stale[bot]

sherif-elmetainy, 2021-02-10:
In had this problem and solved it by running:
`sudo apt install wiringpi`

I am using raspbian.

PeterHric, 2022-05-30:
> sudo apt install wiringpi

Sherif, did you install it on you Pi, or inside the container ? Because the issue deals with Docker container, I believe..

sherif-elmetainy, 2022-05-31:
> > sudo apt install wiringpi
> 
> Sherif, did you install it on you Pi, or inside the container ? Because the issue deals with Docker container, I believe..

Oh, I missed the part about docker when I read and replied to this issue. My apologies. My case was on a physical raspberry.