unosquare/raspberryio #230: "StartSoftPwm" fails on second call

PsBernie, 2020-11-20:
Based on the code described in the documentation, I run

                const int range = 100;
                GpioPin pin = (GpioPin)Pi.Gpio[BcmPin.Gpio18];
                pin.PinMode = GpioPinDriveMode.Output;
                pin.StartSoftPwm(0, range);

                //while (true)
                //{
                for (int x = 0; x <= 100; x++)
                {
                    pin.SoftPwmValue = range / 100 * x;
                    Thread.Sleep(10);
                }

                for (int x = 0; x <= 100; x++)
                {
                    pin.SoftPwmValue = range - (range / 100 * x);
                    Thread.Sleep(10);
                }
                //}

which works fine during the first call.

When I call it again, I receive an error message:
[image]

I thought I may have to add a call to StopSoftPwm afterwards, but there's no such function.
Also, there seems to be no way of finding out if StartSoftPwm has been called before.

What can I do to prevent this error, except by handling some mutable global state in my application?

geoperez, 2021-02-15:
We are looking for a new maintainer, apply at https://adoptoposs.org/p/9f5b74b9-04f2-42b6-891f-c5294c9ef1c5

canadrian72, 2021-03-01:
I also encounter the same issue. StartSoftPwm fails on second call for me as well. Hopefully someone can clairfy how to mitigate this issue.

RickyYChan, 2021-03-25:
I'm a noob to PWM or this kind of things(signal or sth), but i do drive my servo use pwm like this:
```python
import RPi.GPIO as GPIO
import time

servoPIN = 17 # gpio
GPIO.setmode(GPIO.BCM)
GPIO.setup(servoPIN, GPIO.OUT)

p = GPIO.PWM(servoPIN, 50) # gpio and 50 hz of frec
p.start(2.5) # Initialization, duty cycle = 2.5%
try:
  while True:
    p.ChangeDutyCycle(5)
    time.sleep(0.5)
    p.ChangeDutyCycle(7.5)
    time.sleep(0.5)
    p.ChangeDutyCycle(10)
    time.sleep(0.5)
    p.ChangeDutyCycle(12.5)
    time.sleep(0.5)
    p.ChangeDutyCycle(10)
    time.sleep(0.5)
    p.ChangeDutyCycle(7.5)
    time.sleep(0.5)
    p.ChangeDutyCycle(5)
    time.sleep(0.5)
    p.ChangeDutyCycle(2.5)
    time.sleep(0.5)
except KeyboardInterrupt:
  p.stop()
  GPIO.cleanup()
```
in this slice of code, apparently, what i did is:
1. set mode bcm
2. setup pin
3. start with 0
4. rotate from 0° to 180°( duty cycle 2.5% - 12.5%)

in the C# code, i think this is a correct way(but i havent tested yet)
```C#
    class Program
    {
        static void Main(string[] args)
        {
            Console.WriteLine(Pi.Info + Environment.NewLine + Environment.NewLine);
            Pi.Init<BootstrapWiringPi>();

            // GPIO.setmode(BCM), 
            var servo = (GpioPin)Pi.Gpio[BcmPin.Gpio12];
            // GPIO.setup(servo, GPIO.OUT)
            servo.PinMode = GpioPinDriveMode.Output;
            // set duty cycle maximum.
            // pwmrange = 100% duty cycle.
            servo.PwmRange = 1000;
            // pwmregister is similar to duty cycle now.

            // to be precision:
            // pwmregister / pwmrange = duty cycle.
            servo.StartSoftPwm(0, 25);
            // duty cycle 2.5 = angle 0
            // duty cycle 12.5 = angle 180
            // per duty cycle = angle 18
            try
            {
                while (true)
                {
                    servo.PwmRegister = 50;
                    Thread.Sleep(500);
                    servo.PwmRegister = 75;
                    Thread.Sleep(500);
                    servo.PwmRegister = 100;
                    Thread.Sleep(500);
                    servo.PwmRegister = 125;
                    Thread.Sleep(500);
                    servo.PwmRegister = 100;
                    Thread.Sleep(500);
                    servo.PwmRegister = 75;
                    Thread.Sleep(500);
                    servo.PwmRegister = 50;
                    Thread.Sleep(500);
                    servo.PwmRegister = 25;
                    Thread.Sleep(500);
                }
            }
            catch (Exception e)
            {
                Console.WriteLine(e.Message);
            }
        }
    }
```
so if i wanna translate my python code to C# code, i still have 2 things to do:
1. stop the servo
2. cleanup the gpio like python did
If i donnot worked like that, the python will throw an exception which tells me that my servo has been called or busy or sth.(i dont remember clearly)
So the solution is to find the stop method and cleanup method in the RaspberryIO, but i donnot know where it is ...

RickyYChan, 2021-03-25:
> I'm a noob to PWM or this kind of things(signal or sth), but i do drive my servo use pwm like this:
> 
> ```python
> import RPi.GPIO as GPIO
> import time
> 
> servoPIN = 17 # gpio
> GPIO.setmode(GPIO.BCM)
> GPIO.setup(servoPIN, GPIO.OUT)
> 
> p = GPIO.PWM(servoPIN, 50) # gpio and 50 hz of frec
> p.start(2.5) # Initialization, duty cycle = 2.5%
> try:
>   while True:
>     p.ChangeDutyCycle(5)
>     time.sleep(0.5)
>     p.ChangeDutyCycle(7.5)
>     time.sleep(0.5)
>     p.ChangeDutyCycle(10)
>     time.sleep(0.5)
>     p.ChangeDutyCycle(12.5)
>     time.sleep(0.5)
>     p.ChangeDutyCycle(10)
>     time.sleep(0.5)
>     p.ChangeDutyCycle(7.5)
>     time.sleep(0.5)
>     p.ChangeDutyCycle(5)
>     time.sleep(0.5)
>     p.ChangeDutyCycle(2.5)
>     time.sleep(0.5)
> except KeyboardInterrupt:
>   p.stop()
>   GPIO.cleanup()
> ```
> 
> in this slice of code, apparently, what i did is:
> 
> 1. set mode bcm
> 2. setup pin
> 3. start with 0
> 4. rotate from 0° to 180°( duty cycle 2.5% - 12.5%)
> 
> in the C# code, i think this is a correct way(but i havent tested yet)
> 
> ```cs
>     class Program
>     {
>         static void Main(string[] args)
>         {
>             Console.WriteLine(Pi.Info + Environment.NewLine + Environment.NewLine);
>             Pi.Init<BootstrapWiringPi>();
> 
>             // GPIO.setmode(BCM), 
>             var servo = (GpioPin)Pi.Gpio[BcmPin.Gpio12];
>             // GPIO.setup(servo, GPIO.OUT)
>             servo.PinMode = GpioPinDriveMode.Output;
>             // set duty cycle maximum.
>             // pwmrange = 100% duty cycle.
>             servo.PwmRange = 1000;
>             // pwmregister is similar to duty cycle now.
> 
>             // to be precision:
>             // pwmregister / pwmrange = duty cycle.
>             servo.StartSoftPwm(0, 25);
>             // duty cycle 2.5 = angle 0
>             // duty cycle 12.5 = angle 180
>             // per duty cycle = angle 18
>             try
>             {
>                 while (true)
>                 {
>                     servo.PwmRegister = 50;
>                     Thread.Sleep(500);
>                     servo.PwmRegister = 75;
>                     Thread.Sleep(500);
>                     servo.PwmRegister = 100;
>                     Thread.Sleep(500);
>                     servo.PwmRegister = 125;
>                     Thread.Sleep(500);
>                     servo.PwmRegister = 100;
>                     Thread.Sleep(500);
>                     servo.PwmRegister = 75;
>                     Thread.Sleep(500);
>                     servo.PwmRegister = 50;
>                     Thread.Sleep(500);
>                     servo.PwmRegister = 25;
>                     Thread.Sleep(500);
>                 }
>             }
>             catch (Exception e)
>             {
>                 Console.WriteLine(e.Message);
>             }
>         }
>     }
> ```
> 
> so if i wanna translate my python code to C# code, i still have 2 things to do:
> 
> 1. stop the servo
> 2. cleanup the gpio like python did
>    If i donnot worked like that, the python will throw an exception which tells me that my servo has been called or busy or sth.(i dont remember clearly)
>    So the solution is to find the stop method and cleanup method in the RaspberryIO, but i donnot know where it is ...

Unfortunately, I cannot use this package, so i decided to use Python to drive my servo. :)

closed 2021-06-16 by stale[bot]