unosquare/raspberryio #231: How to Use ADS1115 I2C Analog-to-Digital Converter devices

DevilK89, 2020-12-02:
Good afternoon,

I would like to know if you have any example of communication with the ADS1115 card, as I am not able to communicate with the card, currently it appears the error id 72 could not be registered with the I2C bus.

Thank you for now.

multisys, 2020-12-17:
Hi, I am having the same problem.

multisys, 2020-12-17:
**DevilK89** please put your code.

DerpyDoom, 2020-12-28:
This is the way I'm using the ADS1115:

```
var mainADC = new Unosquare.RaspberryIO.Peripherals.ADS1115();

while (true)
{
    Console.WriteLine("Raw Voltage: " + mainADC.ReadChannel(0x0)); //Read Channel 1
    Console.WriteLine("Raw Voltage: " + mainADC.ReadChannel(0x1)); //Read Channel 2
    Console.WriteLine("Raw Voltage: " + mainADC.ReadChannel(0x2)); //Read Channel 3
    Console.WriteLine("Raw Voltage: " + mainADC.ReadChannel(0x3)); //Read Channel 4
    Thread.Sleep(1000);
}
```

geoperez, 2021-02-15:
We are looking for a new maintainer, apply at https://adoptoposs.org/p/9f5b74b9-04f2-42b6-891f-c5294c9ef1c5

closed 2021-06-16 by stale[bot]