unosquare/raspberryio #241: HRESULT: 0x8007007E

Nickellukas2001, 2021-11-22:
Hey, so i recently tried to set up unosquare to my c# project in Visual Studio. I have just started with programming so i dont know much and really need help here.

1. Is there some kind of a "Dictionary" whre i can find all the commands and how to use them? Would be great.
2. When I started the Program first it said: "You must first initialize Pi referencing a valid IBootstrap implementation."So i included unosquare.WiringPi and wrote into my public Form1( )             "Pi.Init<BootstrapWiringPi>();". But when i start the Program it gives me the error: "The DLL" libc ": The specified module was not found. (Exception from HRESULT: 0x8007007E) cannot be loaded."

So i have no idea how to fix that and where it is coming from. It would be really nice if someone could help me with that.

Greetings,
Lukas
[image]

Nickellukas2001, 2021-11-24:
@geoperez i saw that you are a member of @Unosquare ,do you maybe have an idea?

geoperez, 2021-11-24:
Hi, the library will only work in Linux since it's a wrapper around the WiringPi lib. You need to create a console app and run it directly in your raspberry PI.

Nickellukas2001, 2021-11-24:
oh okay, so i just write my code and then create the app and run it with mono on the raspberry PI? And aboout the functions of Unosquare: is there some kind of "Dictionary" where i can find alöl the commands and how to use them?

Nickellukas2001, 2021-11-25:
@geoperez does this mean the lib is only for console Applications and cant be used in Windows forms applications run with mono on the pi?

geoperez, 2021-11-28:
Windows Forms are not available in Mono. You can use GTK# or similar UI frameworks.

Nickellukas2001, 2021-11-28:
Hi, thanks for your help. I already found an different solution. Now I'm using system.device.gpio as a library and it works perfect. But thanks again for the help.

closed 2021-11-28 by Nickellukas2001